UMSAE-Formula-Electric/MECH #16: October 16th Rolling Chassis

HVTwisty, 2025-09-18:
## Summary

This serves as a snapshot of the open items needed to be completed before we have a rolling chassis.

@UMSAE-Formula-Electric/mechanical-leads  comment all open items preventing us from having a rolling chassis below. If you do not have any items blocking us from rolling chassis comment "None"

HaydenMorin12, 2025-09-19:
Accumulator:

No current rolling chassis restrictions

Update

-Fits in the frame (Finally)

-Figured out good way apply the epoxy to the cells (will be working on this more this weekend)

-LV needs to be brought up to speed eventually

-Charge cart will be left until later

AngeloGrandePT, 2025-09-19:
Powertrain:

No current rolling chassis restrictions.

(I don't believe the half shafts and diff need to be on for the wheels to roll?)

Jessica112334, 2025-09-19:
none